PricelessToolkit/MailBoxGuard #2: Boom file

danielducu, 2023-08-19:
I think boom file is incorrect / incomplete.

PricelessToolkit, 2023-08-19:
Why?

danielducu, 2023-08-20:
> Why?

More of a request then an issue hehe, the "Supplier and ref" is missing.

PricelessToolkit, 2023-08-21:
You can buy components wherever you want Digikey.com, lcsc.com.... right now it's hard to find components in one place.

PricelessToolkit, 2023-08-21:
Added manufacturer's part numbers for important parts only. for resistors and LEDs there are sizes and values.

CommittotheIndian, 2023-12-15:
I've attempted to upload this to PCBway and have them assemble. It would appear they are having issues with your formatting. The assembly step has been in audit for hours with no response from PCBway.

Any suggestions?

PricelessToolkit, 2023-12-15:
I think they need to check components' stock disponibility..... so wait until the audit ends "It can take 24+ hours"

CommittotheIndian, 2023-12-16:
They are having issues with designators and part numbers on the last 3 items. 

They also asked about the battery but I can purchase separate.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:////Users/PunkButt/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip.htm">
<link rel=File-List
href="file:////Users/PunkButt/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip_filelist.xml">
<style>

</style>
</head>

<body link=blue vlink=purple>

30 |   | 1 |   | Fresnel lens   for AS312 | Aliexpress
-- | -- | -- | -- | -- | --
31 |   | 1 |   | 433/868/915Mhz Spring   Antenna90d | Aliexpress
32 |   | 1 | MY-AA-04 | MYOUNG MY-AA-04 AA   Battery Holder | C2979179

</body>

</html>

[T-B2W582299A-5sets-MailBox_Guard_Boom(2023-12-16).xls](https://github.com/PricelessToolkit/MailBoxGuard/files/13693013/T-B2W582299A-5sets-MailBox_Guard_Boom.2023-12-16.xls)

PricelessToolkit, 2023-12-16:
30,31 You need to buy from Ali and hand solder, 32 from LCSC need to be hand-soldered on the back side of the PCB.   PS: how much do they want for full assembly + shipping? is the difference very big? compared to my store?

CommittotheIndian, 2023-12-16:
I know those need to be hand soldered but hoping if I just added that note, they could just throw them in the box (save me on cost since I don't need batches of 10+).

I started this with 5 units but I only need 1. They are the same cost 1 vs 5 accept when it comes to the individual parts. So If I take it back down to 1 assembled and shipped, I think I'm looking at $48-$50 US. Right now I'm just getting quotes.

I'd much rather just purchase from you since it is your design and I like helping the little guy but with yours + shipping + and EU conversion, it's $70 just for one unit without anything else. I was going to spend over $100 for a mailbox sensor.

PricelessToolkit, 2023-12-16:
868Mhz version tested + shipping is 65.45 USD / 433 is 61 USD. For me no difference whether you pay for an assembly service or buy a directly from my shop. The main thing for me is that you liked the project! if you have any questions about the setup, open a discussion, good luck 👍

closed 2023-12-28 by PricelessToolkit