PricelessToolkit/MailBoxGuard #6: Request: OTAA

WhistleMaster, 2023-08-23:
Hi !

Thanks again the great work on those projects! 

I was wondering if that would be possible to have a version of the sensor code which would use OTAA to join other Lora networks such as TTN ?

Thanks !
Kind regards,

WM

PricelessToolkit, 2023-08-23:
Hi!

I think there is only one gateway in my area and it is far away, so I have never tried using TTN. But if I find free time, I'll try. Do you have experience with Arduino coding?

WhistleMaster, 2023-08-23:
Thanks ! I have a bit of experience with Arduino coding. I've found some example such as [MailboxNotifier](https://github.com/SensorsIot/MailboxNotifier/blob/main/MailboxNotifier1284P_V3/MailboxNotifier1284P_V3.ino), similar to the MaiBoxGuard sensor, but using OTAA to join LoraWan networks or also [LilyGo](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/LoRaWAN/OTTA/OTTA.ino).

WhistleMaster, 2023-08-23:
Based on those examples, I've tried something: [ttn_otaa.ino.zip](https://github.com/PricelessToolkit/MailBoxGuard/files/12421591/ttn_otaa.ino.zip) but the issue I have is, I guess, the limited size of the micro controller Attiny1616 because I get an error at compilation which is:

```
region `text' overflowed by 9345 bytes
```

PricelessToolkit, 2023-08-23:
can you attach libraries too?

WhistleMaster, 2023-08-24:
Sure ! The only additional one needed is `MCCI_LoRaWAN_LMIC_library` [libraries.zip](https://github.com/PricelessToolkit/MailBoxGuard/files/12425635/libraries.zip)

PricelessToolkit, 2023-08-24:
Error compiling for board ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406.
Library not working with Attiny1616

WhistleMaster, 2023-08-24:
This is where my competences with Arduino coding stop I guess. I'll leave you mastering it better than I could do 😄

PricelessToolkit, 2023-08-25:
I'm more into hardware than software so nop! 😊

WhistleMaster, 2023-08-31:
I've found another lib from Adafruit which could be good for OTAA. I'd need the following PINs. 
- The Lora module's interrupt pin (rfm_nss).
- The Lora module's slave select pin (rfm_nss).
- The Lora module's reset pin (rfm_rst).

I had a look at the schematic but I'm not sure which pin of the Ai-Thinker-Ra you used.

Thanks !

PricelessToolkit, 2023-08-31:
ARDUINO | ATTINY | LORA

GPIO14 - PA1 - MOSI
GPIO15 - PA2 - MISO
GPIO16 -  PA3 - SCK/CLK
GPIO1 -  PA5 - RESET
GPIO10 - PC0 - NSS

WhistleMaster, 2023-09-01:
Thanks ! 

I have the NSS and the RST pins, but I still don't get what the interrupt pin is. Should be DI0 on the LoRa chip but to what pin on the ATTINY is it connected ?

[image]

PricelessToolkit, 2023-09-01:
DI0 is not connected, Since the sensor does not have a sleep mode "power is completely turned off" DI0 is not used. DI0 is used for two-way comunication "if I understand correctly". So DI0 is not required.

PricelessToolkit, 2023-09-01:
"The dio0 pin can be used for channel activity detection callback, transmission finish callback and/or receiving callback, check onCadDone , onTxDone, and onReceive."

WhistleMaster, 2023-09-14:
So what would be the module's interrupt pin ?

WhistleMaster, 2023-10-01:
Is this closed because the sensor has now the OTAA function ?

PricelessToolkit, 2023-10-01:
I closed it because of inactivity + the interrupt pin is not connected to the LoRa module.

WhistleMaster, 2023-10-17:
Well, we can keep it open, just in case someone is interested to jump in to help :)

PricelessToolkit, 2023-10-17:
ok

WhistleMaster, 2023-10-18:
I've done other tests and unfortunately the TinyLora lib from Adafruit is not compatible with TTNv3... and it seems that there is no other lib working with the Attiny1616...

PricelessToolkit, 2023-10-18:
Hi, if i understand correctly TTN uses one-way communication, then why all TTN libs require DI0 I can't understand.

WhistleMaster, 2023-10-19:
It seems to be related to single-channel packet forwarders that no longer work after the Things Network migration to The Things Stack v3. according to that forum thread: https://www.thethingsnetwork.org/forum/t/single-channel-packet-forwarders-scpf-are-deprecated-and-not-supported/31117

WhistleMaster, 2023-12-05:
I found the following project https://github.com/novag/LoRa-ATtiny-Node, for ATtiny85, which implements the ABP and OTAA. 

Maybe that could also work with the Attiny1616 ?

PricelessToolkit, 2023-12-06:
Autor says "Currently this library is heavily optimized for ATtiny85 chips and needs changes to run on other devices" + "It has support for downlink messages" it uses "DI0" too,  so I don't think it will work.

WhistleMaster, 2023-12-09:
That's a pity... I found somewhere that you could also achieve the OTAA by sending AT commands, such as the following:

```
AT+CJOINMODE=0
AT+CRXP=1,1,869525000
AT+CDEVEUI=70B3D57ED0052256
AT+CAPPEUI=0000000000000000
AT+CAPPKEY=8D5EC8F6834BDE677A3D815E81A14981
AT+CFREQBANDMASK=0001
AT+CULDLMODE=2
AT+CJOIN=1,1,8,8
AT+DTRX=0,2,6,445566
AT+DTRX=1,1,4,5555
```

Is that something that could be supported by the lora chip you use ?

PricelessToolkit, 2023-12-09:
Hi, unfortunately, no! These AT commands for assemblies like the "MCU+LoRa"  stand-alone LoRa chip can't support AT commands, "it does not speak TTL"

itopaloglu83, 2024-01-29:
I was about to ask about the same The Things Network (TTN) support when I found this issue. It is quite unfortunate that we are not able to make it work.  Thank you for all the research and experimentation. 

Could you provide a summary of what the problem is? I'm not very familiar with the requirements for TTN so I would like to research it a little bit more. The form link about talks about packet forwarders but my understanding is that MailBoxGuard would be an end node only.

PricelessToolkit, 2024-01-29:
The problem is that this sensor was not designed to work with TTN, "I didn't know that to work with TTN" I needed to connect the DI0 LoRa pin to an interrupt-capable MCU pin. "it is for detecting when the transmission is done" TTN libraries require this connection, but I don't know why....

itopaloglu83, 2024-01-30:
First of all, you did a great job with the board and the video. I already bought one and now experimenting to see if I can get away with not having a gateway and use TTN instead. It's a fun side project to tinker with.

Arduino LoRa library says that the DIO0 is optional and it is only needed for the receive callback mode. I also remember watching Andreas Spiess' video about the MailboxNotifier, he used the interrupt to keep the MCU sleeping while the packets were being sent. 

I'm going to read the sample codes above and see if there's any walk around.

itopaloglu83, 2024-02-02:
This update is for any future curious souls. After experimenting with an OTAA capable board for a while and realizing that there is no TTN gateway close to where I live, I abandoned figuring out how to connect MailboxGuard to TTN.

PricelessToolkit, 2024-02-02:
> This update is for any future curious souls. After experimenting with an OTAA capable board for a while and realizing that there is no TTN gateway close to where I live, I abandoned figuring out how to connect MailboxGuard to TTN.

Thanks for trying, I don't have a close gateway either.

WhistleMaster, 2024-02-02:
I do have a gateway and would be really happy to test !

itopaloglu83, 2024-02-02:
I don't have an exact procedure to follow but here's the approach I was planning to take. 

Practically speaking, the interrupt functionality is only needed when the LoRa Radio module is receiving a package and needs to inform the microcontroller about it. This is required because during OTAA setup, various back-and-forward communication takes place to exchange encryption keys etc. One could theoretically make a small sketch that registers to the TTN network by sending the required messages (APPID etc.) and then polling for a response from the radio instead of using the interrupts. In a way, writing an alternative LoRaWan library that can work on AtTiny1616. 

If you just want to experiment without writing your own LoRaWan library, you can try something like [Dragino LDS02](https://www.dragino.com/products/lorawan-nb-iot-door-sensor-water-leak/item/181-lds02.html) that is already capable of joining TTN via OTAA.

WhistleMaster, 2024-02-05:
Yes, indeed, we could use another sensor from Dragino but it would be better to use the MailBoxGuard 🙂

PricelessToolkit, 2024-06-22:
All will be fixed with the next HW version.

closed 2024-06-22 by PricelessToolkit