PricelessToolkit/MailBoxGuard #17: Error when compiling

Gahrlaag, 2023-10-01:
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Aleks\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Aleks\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Aleks\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=1616,clock=5internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,PWMmux=A_default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Aleks\AppData\Local\Temp\arduino_build_809065 -warnings=none -build-cache C:\Users\Aleks\AppData\Local\Temp\arduino_cache_912473 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.python3.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -verbose C:\Users\Aleks\Desktop\MailBoxGuard-main\Code\Arduino\Mailbox_Guard_Sensor\Mailbox_Guard_Sensor.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Aleks\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Aleks\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Aleks\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy6:chip=1616,clock=5internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=enabled,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,PWMmux=A_default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Aleks\AppData\Local\Temp\arduino_build_809065 -warnings=none -build-cache C:\Users\Aleks\AppData\Local\Temp\arduino_cache_912473 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino6.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino6 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.avrdude-6.3.0-arduino17or18.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18 -prefs=runtime.tools.python3.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Aleks\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1 -verbose C:\Users\Aleks\Desktop\MailBoxGuard-main\Code\Arduino\Mailbox_Guard_Sensor\Mailbox_Guard_Sensor.ino
Using board 'atxy6' from platform in folder: C:\Users\Aleks\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.8
Using core 'megatinycore' from platform in folder: C:\Users\Aleks\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.8
Detecting libraries used...
"C:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1616 -DF_CPU=5000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10813 -DARDUINO_AVR_ATtiny1616 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.8.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=8UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy6 "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\cores\\megatinycore" "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\variants\\txy6" "C:\\Users\\Aleks\\AppData\\Local\\Temp\\arduino_build_809065\\sketch\\Mailbox_Guard_Sensor.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for SPI.h: [SPI@1.1.2]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.1.2]
"C:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1616 -DF_CPU=5000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10813 -DARDUINO_AVR_ATtiny1616 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.8.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=8UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy6 "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\cores\\megatinycore" "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\variants\\txy6" "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\libraries\\SPI\\src" "C:\\Users\\Aleks\\AppData\\Local\\Temp\\arduino_build_809065\\sketch\\Mailbox_Guard_Sensor.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for LoRa.h: [LoRa@0.8.0]
ResolveLibrary(LoRa.h)
  -> candidates: [LoRa@0.8.0]
"C:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny1616 -DF_CPU=5000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERD0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10813 -DARDUINO_AVR_ATtiny1616 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.8.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=8UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy6 "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\cores\\megatinycore" "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\variants\\txy6" "-IC:\\Users\\Aleks\\AppData\\Local\\Arduino15\\packages\\megaTinyCore\\hardware\\megaavr\\2.6.8\\libraries\\SPI\\src" "-IC:\\Users\\Aleks\\Documents\\Arduino\\libraries\\LoRa\\src" "C:\\Users\\Aleks\\AppData\\Local\\Temp\\arduino_build_809065\\sketch\\Mailbox_Guard_Sensor.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
[... cut ...]
Mailbox_Guard_Sensor_MQTT_Auto_Discovery:53:6: error: redefinition of 'void loop()'
 void loop() {
      ^~~~
C:\Users\Aleks\Desktop\MailBoxGuard-main\Code\Arduino\Mailbox_Guard_Sensor\Mailbox_Guard_Sensor.ino:52:6: note: 'void loop()' previously defined here
 void loop() {
      ^~~~
Using library SPI at version 1.1.2 in folder: C:\Users\Aleks\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.8\libraries\SPI 
Using library LoRa at version 0.8.0 in folder: C:\Users\Aleks\Documents\Arduino\libraries\LoRa 
exit status 1
redefinition of 'String NewMailCode'

PricelessToolkit, 2023-10-01:
Hi, 
1. Arduino IDE version?
2. Download "Mailbox_Guard_Sensor_HARestAPI.ino" don't just copy past it

Gahrlaag, 2023-10-01:
1. 2.2.1
2. I haven't copypasted anything. I downloaded the entire zip from the main github, including "Mailbox_Guard_Sensor_HARestAPI.ino" and "Mailbox_Guard_Sensor.ino"

PricelessToolkit, 2023-10-01:
https://github.com/PricelessToolkit/MailBoxGuard#arduino-setup-ide-20x-unsupported

Gahrlaag, 2023-10-01:
> https://github.com/PricelessToolkit/MailBoxGuard#arduino-setup-ide-20x-unsupported

Not following? All of the libraries are present and the correct board is selected.

PricelessToolkit, 2023-10-01:
Arduino "IDE 2.0.x is unsupported, its a "megaTinyCore" Recommendation

PricelessToolkit, 2023-10-01:
I am using 1.8.19 and the code compiles without any problems

Gahrlaag, 2023-10-01:
I just tried 1.8.57.0 and I get the same error. Exact same settings

Gahrlaag, 2023-10-01:
Same with 1.8.19

PricelessToolkit, 2023-10-01:
Do you modify the code?

I think it is a TAB or ESPACE issue try highlighting all the text in the IDE
Select Tools > AutoFormat

PricelessToolkit, 2023-10-01:
you get errors like this 

`loop()' previously defined here
void loop() {
^~~~`

only if you messed up with the formatting text

Gahrlaag, 2023-10-01:
> Do you modify the code?
> 
> I think it is a TAB or ESPACE issue try highlighting all the text in the IDE Select Tools > AutoFormat

I havent touched the code aside from changing the values of NewMailCode and LowBatteryCode. 

here's a pastebin of the exact code

https://pastebin.com/cA2nHuq5

PricelessToolkit, 2023-10-01:
Pastebin code compiled widout problem. you have problem with your Arduino installation

Gahrlaag, 2023-10-01:
> Pastebin code compiled widout problem. you have problem with your Arduino installation

I've tried it on two different computers :D WTF....

Gahrlaag, 2023-10-01:
Is there a library missing? prerequisites?

PricelessToolkit, 2023-10-01:
only these
#include <SPI.h>
#include <LoRa.h>
#include <avr/sleep.h>

did you install the libraries or use those from the project?

Gahrlaag, 2023-10-01:
I installed them.. should I use the ones from the project?

PricelessToolkit, 2023-10-01:
I am using libraries from the project

PricelessToolkit, 2023-10-01:
and one stupid question, in the sketch folder is there only one ".ino" file?

Gahrlaag, 2023-10-01:
> and one stupid question, in the sketch folder is there only one ".ino" file?

Goddammit I'm stupid... No, of course not. 
Compiles as it should now... D'oh

closed 2023-10-01 by PricelessToolkit