PricelessToolkit/MailBoxGuard #18: Inject 3.3v to an unpowered LDO output?

pavel-d, 2023-10-09:
Hey @PricelessToolkit, I'm curious if there are any risk when powering the board directly via 3.3v programming pad. The SPX3819's output is directly connected to 3.3v rail (there are no diodes or other components in between), so it will have a 3.3v on the OUT, and 0v on the input. Are there any risk damaging the LDO? I haven't found anything in the datasheet unfortunately

PricelessToolkit, 2023-10-09:
Hi, yes, you can, when you are programming MailBox Guard you doing exactly the same thing.

pavel-d, 2023-10-09:
There is an opinion on the internet that not all LDOs like that and I wasn’t able to find any evidence that the SPX3819 handles it. But it’s good to know that it works fine!