PricelessToolkit/MailBoxGuard #19: question regarding setup

steuerzentrale, 2023-10-13:
first of all many thanks for this nice piece of hardware, it is EXACTLY what i have been looking for! I am about to order the mailbox-sensor from your shop and already looked on how to configure everything. I have some knowledge on ESPHome but only a very basic understanding of Arduino IDE. lets come to my question.

I configured LoRa_Gateway_MQTT.ino with my credentials (Wifi and MQTT) as well as LoRa_Gateway_HARestAPI.ino (with the token). I dont understand what to do with the sketches afterwards.

Do I have to combine them in order to upload it on the TTGO board or do I have to upload the LoRa_Gateway_MQTT.ino first and then LoRa_Gateway_HARestAPI.ino?

Sorry for my noob-question, I couldnt find an answer in the readme.

PricelessToolkit, 2023-10-13:
Hi, do you want to use the MQTT version or RestAPI?  

If MQTT then 
`"Mailbox_Guard_Sensor_MQTT_Auto_Discovery.ino" ` goes into the sensor,

`"LoRa_Gateway_MQTT_Auto_Discovery.ino" and "board.h"` into Gateway

steuerzentrale, 2023-10-13:
thanks for the quick reply, I want to use MQTT. Now I get it, i thought I had to use LoRa_Gateway_MQTT.ino AND LoRa_Gateway_HARestAPI.ino but yeah, at the time of writing I understand that the sensors infos pass through MQTT which is running on my HA-instance.

closed 2023-10-13 by steuerzentrale