PricelessToolkit/MailBoxGuard #39: PIR Sensitivity + Door contact sensor

RubenWaterman, 2024-01-25:
First of all, awesome project and thanks very much! I've got it successfully integrated in my Loxone smart home 🤩
[image]

For some reason, the on-board PIR triggers way too often, when there is no mail inserted. Is there a configuration flag to make the PIR less sensitive?

I was also thinking of soldering a door contact on board (and either attach on the slat that opens when mail is inserted to trigger the "new mail" activity, OR to put the door contact on the actual door so that would trigger the "I've seen the mail" event). I found the correct pins for this on the board but I didn't find any code on how to set this up. Do you happen to have an example for this?

RubenWaterman, 2024-01-25:
Ah, is it possible to trigger two separate types of MQTT messages, i.e. one for PIR activity and one for door contact sensor?

WhistleMaster, 2024-01-25:
I don't think so. The PIR and the door contact switch are both used as trigger on the board.

PricelessToolkit, 2024-01-25:
Thank you for supporting my work.
"For some reason, the on-board PIR triggers way too often" If you have a mailbox with a small slit for mail and it stays open, change the location of the sensor so that the sensor faces in the other direction.

closed 2024-02-10 by PricelessToolkit