PricelessToolkit/MailBoxGuard #49: Screen is always blank

dmfrey, 2024-03-26:
I've uploaded numerous times your lora gateway code, specifically, the MQTT HA ones. Should I see anything on the screen after it uploads? I can see the device on the network.

Also, side note, the MQTT user and password, should they be the HA MQTT User and Password?  I ended up creating a new user for this specifically. But I'm guessing it should be the HA one.

dmfrey, 2024-03-26:
Regarding my side note, I updated it to the HA MQTT User and Password and now I see the entities in HA. That can be disregarded.

PricelessToolkit, 2024-03-26:
Hi, Only the ( LoRa_Gateway_OLED "Standalone" ) firmware utilizes OLED.

dmfrey, 2024-03-26:
@PricelessToolkit Ah, makes sense. I guess I assumed it was gonna do something. I was able to get it up and running an HA recognized it. Now I'm just waiting on the sensor to try it out :)

dmfrey, 2024-03-26:
closes #49

closed 2024-03-26 by dmfrey