PricelessToolkit/MailBoxGuard #50: VDD was not declared in this scope

dmfrey, 2024-04-16:
Please bare with me, i'm new to arduino

I'm attempting to build the Mailbox_Guard_Sensor and I'm getting the following error:

```
Arduino: 1.8.19 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, Core 1, Core 1, Debug, Disabled"

/Users/dfrey/Development/git/PricelessToolkit/MailBoxGuard/Code/Arduino/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor.ino: In function 'void setup()':
Mailbox_Guard_Sensor:29:19: error: 'VDD' was not declared in this scope
   analogReference(VDD);
                   ^~~
Mailbox_Guard_Sensor:29:3: error: 'analogReference' was not declared in this scope
   analogReference(VDD);
   ^~~~~~~~~~~~~~~
/Users/dfrey/Development/git/PricelessToolkit/MailBoxGuard/Code/Arduino/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor.ino:29:3: note: suggested alternative: 'analogGetChannel'
   analogReference(VDD);
   ^~~~~~~~~~~~~~~
   analogGetChannel
/Users/dfrey/Development/git/PricelessToolkit/MailBoxGuard/Code/Arduino/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor.ino: In function 'void loop()':
Mailbox_Guard_Sensor:54:31: error: 'PIN_PB4' was not declared in this scope
   float volts = analogReadEnh(PIN_PB4, 12)*(1.1/4096)*(30+10)/10;
                               ^~~~~~~
/Users/dfrey/Development/git/PricelessToolkit/MailBoxGuard/Code/Arduino/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor.ino:54:31: note: suggested alternative: 'PIN_DAC2'
   float volts = analogReadEnh(PIN_PB4, 12)*(1.1/4096)*(30+10)/10;
                               ^~~~~~~
                               PIN_DAC2
Mailbox_Guard_Sensor:54:17: error: 'analogReadEnh' was not declared in this scope
   float volts = analogReadEnh(PIN_PB4, 12)*(1.1/4096)*(30+10)/10;
                 ^~~~~~~~~~~~~
/Users/dfrey/Development/git/PricelessToolkit/MailBoxGuard/Code/Arduino/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor.ino:54:17: note: suggested alternative: 'analogRead'
   float volts = analogReadEnh(PIN_PB4, 12)*(1.1/4096)*(30+10)/10;
                 ^~~~~~~~~~~~~
                 analogRead
Mailbox_Guard_Sensor:87:20: error: 'SLEEP_MODE_PWR_DOWN' was not declared in this scope
     set_sleep_mode(SLEEP_MODE_PWR_DOWN);
                    ^~~~~~~~~~~~~~~~~~~
Mailbox_Guard_Sensor:87:5: error: 'set_sleep_mode' was not declared in this scope
     set_sleep_mode(SLEEP_MODE_PWR_DOWN);
     ^~~~~~~~~~~~~~
Mailbox_Guard_Sensor:88:5: error: 'sleep_enable' was not declared in this scope
     sleep_enable();
     ^~~~~~~~~~~~
/Users/dfrey/Development/git/PricelessToolkit/MailBoxGuard/Code/Arduino/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor.ino:88:5: note: suggested alternative: 'spiSSEnable'
     sleep_enable();
     ^~~~~~~~~~~~
     spiSSEnable
Mailbox_Guard_Sensor:89:5: error: 'sleep_mode' was not declared in this scope
     sleep_mode();  // Now enter sleep mode.
     ^~~~~~~~~~
/Users/dfrey/Development/git/PricelessToolkit/MailBoxGuard/Code/Arduino/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor.ino:89:5: note: suggested alternative: 'sleep'
     sleep_mode();  // Now enter sleep mode.
     ^~~~~~~~~~
     sleep
exit status 1
'VDD' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

I'm guessing it has to do with the board selected, which is currently the lora gateway board. I'm not really sure what to set it to for the universal programmer. A lot of these look like they're functions that haven't been imported in the header.

Thank you for your help.

PricelessToolkit, 2024-04-16:
Hi, 

Select Board type "ATTiny1616 5MHz" https://github.com/PricelessToolkit/MailBoxGuard?tab=readme-ov-file#mailbox-sensor-configuration

dmfrey, 2024-04-16:
Thank you. I'm not seeing that board in the board manager. Is there a specific one I need to install?

PricelessToolkit, 2024-04-16:
You need to install "megaTinyCore" library, it will add ATTiny support. So add this link to Boards Manager "http://drazzy.com/package_drazzy.com_index.json"

https://github.com/PricelessToolkit/MailBoxGuard?tab=readme-ov-file#arduino-setup-ide-20x-unsupported

dmfrey, 2024-04-16:
Thank you for your help. I think I'm getting close to getting the code to verify in the ide. There seems to be one more exception:
```
Arduino: 1.8.19 (Mac OS X), Board: "ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406, ATtiny1616, 5 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (no reset pin), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), PB0-2, PA3-5, 1-series: PC0/1 (default), On all pins, with new implementation., Disabled (recommended), No delay before window "opens""

/Users/dfrey/Development/git/PricelessToolkit/MailBoxGuard/Code/Arduino/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor/Mailbox_Guard_Sensor.ino: In function 'void loop()':
Mailbox_Guard_Sensor:87:5: error: 'set_sleep_mode' was not declared in this scope
     set_sleep_mode(SLEEP_MODE_PWR_DOWN);
     ^~~~~~~~~~~~~~
Mailbox_Guard_Sensor:88:5: error: 'sleep_enable' was not declared in this scope
     sleep_enable();
     ^~~~~~~~~~~~
Mailbox_Guard_Sensor:89:5: error: 'sleep_mode' was not declared in this scope
     sleep_mode();  // Now enter sleep mode.
     ^~~~~~~~~~
exit status 1
'set_sleep_mode' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

dmfrey, 2024-04-17:
I noticed that this include was commented out:
```
#include <avr/sleep.h>
```
I uncommented it and now it compiles. Not sure if I did that or not.

dmfrey, 2024-04-17:
Just checked the code in the repo and it is commented out. Is that intended? Where does that function come from then? Am I missing an included library somewhere?

dmfrey, 2024-04-17:
I got it working. thank you for your help.

closed 2024-04-17 by dmfrey

PricelessToolkit, 2024-04-17:
> Just checked the code in the repo and it is commented out. Is that intended? Where does that function come from then? Am I missing an included library somewhere?

Hi, I can't explain why it was commented out; it was an error on my part. I will uncomment it. Thanks for pointing it out.