PricelessToolkit/MailBoxGuard #51: Question: Is there a way to set up a second reed switch to reset the counter to zero when opening the mailbox?

ndbroadbent, 2024-04-20:
Hello, thanks for the awesome design! I ordered one a while ago and it's been working very well. I 3D printed a custom flap with magnets for my mailbox, so it triggers a reed switch when a letter opens the flap.

There's just one small feature that I would like to add. It would be great if I could get the mailbox sensor to send a "reset count" message whenever I open the main door on our mailbox to collect the mail. I could add a second reed switch on the main door, but I'm wondering if there's a spare GPIO pin somewhere that I could use for this purpose.

If not, how difficult would it be to update the PCB design? I would be happy to order a second sensor if this feature was added. Thanks a lot!

PricelessToolkit, 2024-04-20:
Hi, thanks for supporting my work!!, no, unfortunately, there is no way to add that functionality to the current version. I am already working on the new version :) and I will make sure to add GPIO pins specially for that purpose.

closed 2024-04-20 by PricelessToolkit

schweighart, 2024-11-04:
I'm thinking about adding a reed switch between the battery and the board. With that it would cut off power whenever the door is opened to collect the mail. Then the board could send the mailbox empty signal in the init block when the power is coming back. The signal from the pir sensor would also need a delay of a few seconds to not trigger the new mail event before the power is cut off. 
I assume I just need a powerful enough reed switch to support the initial power spike. Did I miss something else in my idea?

PricelessToolkit, 2024-11-04:
It won't work because the ATtiny is powered off most of the time, and the trigger relies on physical circuitry. The sensor needs to distinguish between the first activation after power-on and a signal from the PIR sensor, which is impossible with the current hardware version.

schweighart, 2024-11-06:
I totally forgot in my thoughts that the ATtiny is powered down. I somehow had the impression that it is only sleeping. I'm still not willing to give up and came up with 2 other ideas:
1) Cut the connection to the PIR sensor on the back and use a reed sensor instead of the permanent connection. With that the PIR sensor would be cut off as soon as the mailbox door is opened. With the hope that PIR is not faster in triggering.
2) Repurpose the RX or TX connector for the reed sensor. The PIR sensor would still wake up the board but the code could check (probably with a small delay) the state of the reed sensor to know which message to send. The documentation lists these pins also as GPIO but I'm not sure if they would really work as that.

Soldering a tiny wire to a spare pin of the ATtiny seems to be impossible. I also thought about forceful reset the LoRa module from the outside to prevent sending, but I think that would cause other issues without a diode protection the current circuit.

PricelessToolkit, 2024-11-06:
> I totally forgot in my thoughts that the ATtiny is powered down. I somehow had the impression that it is only sleeping. I'm still not willing to give up and came up with 2 other ideas:
>
> 1. Cut the [...]

If you'd like to try a DIY approach, you might want to wait for the next version, which i hope will include three inputs + pir. With those, you'll be able to create more versatile logic setups.

schweighart, 2024-11-09:
No need to wait for the next version :-) . I can confirm that it works. I'll put some instructions here for the next one:
I've soldered my reed sensor (door sensor) between ground and RX (TX would also work).
In the code I've removed all Serial. commands as they would use the same pins.
In the setup block I initialize the pin with: 
`pinMode(PIN_PB3, INPUT_PULLUP); `
(TX would be PIN_PB2). There I also added a 500ms delay to ensure the door magnet is already disconnected when the PIR sensor is triggered.
In the loop part I check if the door is open with the following if statement and I just send an alternative message.
```
if (digitalRead(PIN_PB3)==LOW) {
   //Door closed -> new mail
   LoRa.print(NewMailCode);      
} else {
   //Door open -> mail collected
   LoRa.print(ClearMailCode);      
}
```

PricelessToolkit, 2024-11-10:
Nice !! Do you have 2 doors in your mailbox?

schweighart, 2024-11-10:
No just a normal mailbox with a lid for the postman and a door at the back for me. The board with the PIR sensor is attached with magnets to the back door and is looking to the front lid. 
With that the PIR is waking up on lid movement or back door movement. That's where I needed to distinguish what happend.

schweighart, 2024-11-19:
I had to chance my code a bit. I totally overestimated the time it takes to open the mailbox door in case nothing is in. I'm now checking more often the state of the reed sensor and also removed the delay.

```
#include <SPI.h>
#include <LoRa.h>
#include <avr/sleep.h>

//////////////////////////////////// CONFIG /////////////////////////////////////////////

#define SignalBandwidth 125E3
#define SpreadingFactor 12
#define CodingRate 8
#define SyncWord 0xF3
#define PreambleLength 8
#define TxPower 20
#define BAND 868E6     // frequency in Hz (ASIA 433E6, EU 868E6, US 915E6)
String NewMailCode = "0xA2B2"; // For Example "0xA2B2";
String ClearMailCode = "0xA2B3"; // For Example "0xA2B3";
String LowBatteryCode = "0xLBAT"; // For Example "0xLBAT";

/////////////////////////////////////////////////////////////////////////////////////////

int loopcounter = 0;
boolean doorOpen=false;
//Kept in Memory for about 10 seconds after shutdown. To prevent second sending.
int recentlyTriggered __attribute__((section(".noinit")));;

void setup() {
  pinMode(3, OUTPUT);
  digitalWrite(3, HIGH);
  analogReference(VDD);

  pinMode(PIN_PB2, INPUT_PULLUP); //TX
  pinMode(PIN_PB3, INPUT_PULLUP); //RX
  
  delay(5);

  for (int counter = 0; counter <= 500; counter++) {
    if (digitalRead(PIN_PB3)==HIGH) {
      doorOpen=true;
    }
    delay(10);
  }

  if (!LoRa.begin(BAND)) {
    while (1);
  }

  LoRa.setSignalBandwidth(SignalBandwidth); // signal bandwidth in Hz, defaults to 125E3
  LoRa.setSpreadingFactor(SpreadingFactor); // ranges from 6-12,default 7 see API docs
  LoRa.setCodingRate4(CodingRate);          // Supported values are between 5 and 8, these correspond to coding rates of 4/5 and 4/8. The coding rate numerator is fixed at 4.
  LoRa.setSyncWord(SyncWord);               // byte value to use as the sync word, defaults to 0x12
  LoRa.setPreambleLength(PreambleLength);   //Supported values are between 6 and 65535.
  LoRa.disableCrc();                        // Enable or disable CRC usage, by default a CRC is not used LoRa.disableCrc();
  LoRa.setTxPower(TxPower);                 // TX power in dB, defaults to 17, Supported values are 2 to 20

  if (recentlyTriggered==1) {
    shutdownAll();
  }

  for (int counter = 0; counter <= 500; counter++) {
    if (digitalRead(PIN_PB3)==HIGH) {
      doorOpen=true;
    }
    delay(10);
  }
}

void loop() {
	
  float volts = analogReadEnh(PIN_PB4, 12)*(1.1/4096)*(30+10)/10;
//  Serial.println(volts);
  
  if (loopcounter < 2){
    delay(50);
    LoRa.beginPacket();
    int perc = (int)constrain(((volts - 3.2) / (4.15 - 3.2)) * 100, 0, 100);
    if (doorOpen) {
      //Door open -> mail collected
      LoRa.print(ClearMailCode + "," + perc);      
    } else {
      //Door closed -> new mail
      LoRa.print(NewMailCode + "," + perc);      
    }
    LoRa.endPacket();
    delay (10);
    recentlyTriggered=1;
   }

  if (volts < 3.36 and loopcounter == 1 ){   // Don't change "3.36" !!
    LoRa.beginPacket();
    LoRa.print(LowBatteryCode);
    LoRa.endPacket();
    delay(50);
  }

  if(loopcounter > 2){
    shutdownAll();
  }

  loopcounter ++;
  delay(10);
}

void shutdownAll() {
    LoRa.sleep();  // Put the radio in sleep mode
    digitalWrite(3, LOW); // Sets the Latch pin 3 LOW For power cut off
    
    /* Set sleep mode to PWR_DOWN */
    // There are five different sleep modes in order of power saving:
    // SLEEP_MODE_IDLE - the lowest power saving mode
    // SLEEP_MODE_ADC
    // SLEEP_MODE_PWR_SAVE
    // SLEEP_MODE_STANDBY
    // SLEEP_MODE_PWR_DOWN - the highest power saving mode
    
    set_sleep_mode(SLEEP_MODE_PWR_DOWN);
    sleep_enable();
    sleep_mode();  // Now enter sleep mode.  
}

```