PricelessToolkit/MailBoxGuard #52: Waking up from deep sleep on MailboxGuard does not work

vladimir-aubrecht, 2024-05-03:
Hi,

I bought the mailbox guard, main features are working great, but struggling with waking up the MailboxGuard sensor. Could you please share details on how the device is getting waking up from deep sleep?

My current behaviour is that sensor reports new mail then it goes to deep sleep and does not wake up anymore until I take off battery and put it back (and even that does not work reliably).

PricelessToolkit, 2024-05-03:
HI Vladimir,

The mailbox sensor doesn't sleep at all; when triggered, it powers up the MCU and the LoRa radio. It can be activated by two mechanisms: the PIR sensor or the two contacts ("right side of the name") for the reed switch. The behavior you're describing is unusual.

1: If you don't touch the sensor and only wave your hand in front of the PIR sensor, does the LED turn on?
2: Or does the LED turn on, but you don't receive any signal?

vladimir-aubrecht, 2024-05-03:
Oh I see! I can move my hand or even move with it in any way and it will not turn on the LED 😕 Time to time blue LED is getting on after inserting battery, then it manages to send signal through LoRa and after that it turn off the LED and not reacting again ...

It's using default firmware which was there flashed - I did not touch it.

PricelessToolkit, 2024-05-03:
I think it was damaged during shipping, please contact me through Shop.  and add a link to this issue.

vladimir-aubrecht, 2024-05-03:
I will, thank you! 😊

closed 2024-05-03 by vladimir-aubrecht