PricelessToolkit/MailBoxGuard #57: Error when attempting to compile the mailbox code

mollgal, 2024-09-12:
Hi,
I'm getting an error while attempting to compile the mailbox code Mailbox_Guard_Sensor_Simple.ino, could you please help me figure it out? 

Thanks

The error code I'm getting is:
```
Arduino: 1.8.19 (Linux), Board: "ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406, ATtiny1616, 5 MHz internal, 1.8V (5 MHz or less), Disabled/Disabled, EEPROM retained, Enabled (default timer), UPDI (no reset pin), 8ms, Master or Slave (saves flash and RAM), Default (doesn't print floats, 1.4k flash use), PB0-2, PA3-5, 1-series: PC0/1 (default), On all pins, with new implementation., Disabled (recommended), No delay before window "opens""

Warning: platform.txt from core 'megaTinyCore' contains deprecated compiler.path={runtime.tools.avr-gcc.path}/bin/, automatically converted to compiler.path=/usr/bin/. Consider upgrading this core.
In file included from /home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/Arduino.h:28:0,
                 from sketch/Mailbox_Guard_Sensor_Simple.ino.cpp:1:
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/core_devices.h:293:4: error: #error "No AC? No supported parts exist without one, something is wrong"
   #error "No AC? No supported parts exist without one, something is wrong"
    ^
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/core_devices.h:302:4: error: #error "No ADC? No supported parts exist without one, something is wrong"
   #error "No ADC? No supported parts exist without one, something is wrong"
    ^
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/core_devices.h:351:4: error: #error "No EVSYS detected? All supported parts have one, something is wrong"
   #error "No EVSYS detected? All supported parts have one, something is wrong"
    ^
In file included from /home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/Arduino.h:28:0,
                 from sketch/Mailbox_Guard_Sensor_Simple.ino.cpp:1:
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/core_devices.h:437:4: error: #error "No TCBs? No supported parts exist without one, something is wrong"
   #error "No TCBs? No supported parts exist without one, something is wrong"
    ^
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/core_devices.h:464:4: error: #error "No TWI? No supported parts exist without one, something is wrong"
   #error "No TWI? No supported parts exist without one, something is wrong"
    ^
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/core_devices.h:472:4: error: #error "No SPI? No supported parts exist without one, something is wrong"
   #error "No SPI? No supported parts exist without one, something is wrong"
    ^
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/core_devices.h:493:4: error: #error "No USARTs? No supported parts exist without one, something is wrong"
   #error "No USARTs? No supported parts exist without one, something is wrong"
    ^
In file included from /home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/variants/txy6/pins_arduino.h:27:0,
                 from /home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/Arduino.h:708,
                 from sketch/Mailbox_Guard_Sensor_Simple.ino.cpp:1:
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/timers.h:149:6: error: #error "TCD0, selected for millis, does not exist on this part"
     #error "TCD0, selected for millis, does not exist on this part"
      ^
In file included from /home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/HardwareSerial.h:37:0,
                 from /home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/UART.h:5,
                 from /home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/Arduino.h:838,
                 from sketch/Mailbox_Guard_Sensor_Simple.ino.cpp:1:
/home/user/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.9/cores/megatinycore/UART_check_pins.h:35:4: error: #error "No USARTs present. All supported parts have at least 1 - something is wrong!"
   #error "No USARTs present. All supported parts have at least 1 - something is wrong!"
    ^
exit status 1
Error compiling for board ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

PricelessToolkit, 2024-09-13:
Hi, Please can you check with the latest version of Arduino IDE? I remember someone had exactly the same problem on Linux.

mollgal, 2024-09-13:
I downloaded the latest version (instead of the one from the package manger) and there were no errors in the compilation. 

Thanks a lot!

closed 2024-09-13 by mollgal