PricelessToolkit/MailBoxGuard #60: Compilation error: 'WiFi' was not declared in this scope

leafkicker, 2025-02-25:
When verifying the LoRa_Gateway_HARestAPI.ino code in Arduino I always receive the following error:

`C:\temp\MailBoxGuard-main\Code\Arduino\LoRa_Gateway\LoRa_Gateway_HARestAPI\LoRa_Gateway_HARestAPI.ino: In function 'void setup_wifi()':
C:\temp\MailBoxGuard-main\Code\Arduino\LoRa_Gateway\LoRa_Gateway_HARestAPI\LoRa_Gateway_HARestAPI.ino:42:3: error: 'WiFi' was not declared in this scope
   42 |   WiFi.begin(ssid, password);
      |   ^~~~
C:\temp\MailBoxGuard-main\Code\Arduino\LoRa_Gateway\LoRa_Gateway_HARestAPI\LoRa_Gateway_HARestAPI.ino:44:27: error: 'WL_CONNECTED' was not declared in this scope
   44 |   while (WiFi.status() != WL_CONNECTED) {
      |                           ^~~~~~~~~~~~
Multiple libraries were found for "SSD1306Wire.h"
  Used: C:\Users\Ryan\Documents\Arduino\libraries\ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays
  Not used: C:\Users\Ryan\Documents\Arduino\libraries\SSD1306
exit status 1

Compilation error: 'WiFi' was not declared in this scope`

I have spent hours trying to find this problem, testing with different computers and versions but I am unable to get this to work.

PricelessToolkit, 2025-02-25:
Fixed i added "#include <WiFi.h>" Can i ask why you choiced "LoRa_Gateway_HARestAPI.ino" frimvare ? with that firmware gateway is not universal. consider using "MQTT_JSON"  so you can add other DIY devices too, for more info check capibridge repo

PricelessToolkit, 2025-02-26:
If everything worked, please let me know so I can close this issue.

leafkicker, 2025-02-27:
thankyou it works!

closed 2025-02-27 by leafkicker