PricelessToolkit/MailBoxGuard #64: Disabling the LED and PIR

IDmedia, 2025-05-10:
I just received my mailbox sensor and intend to only use a reed switch with cables. While it's quite easy to see which holes you use for mounting the reed switch, I'm not quite sure how to understand the part about disabling the LED and PIR sensor. In the video you mentioned to "cut through the middle of the jumper pads". I guess that's best done with an x-acto knife? I found it strange that you would have to separate them by cutting instead of rather specifically enabling them by soldering the pads together. Wouldn't that have been much easier?

PricelessToolkit, 2025-05-10:
Yep, just do a couple of passes with an X-Acto knife.

For 99% of people, it's much easier to cut, and every home has a knife. If the LED or PIR were disabled by default, most users wouldn't know if the sensor is working out of the box. This way, it's active by default, and only those who want to disable them (like in your reed switch-only setup) need to make a simple modification.

[image]

IDmedia, 2025-05-10:
Thanks for your quick answer :)
What did you mean by the two pads your marked?

So as far as I understand, for my purpose of only Reed-switch I want to cut at least between the PIR pads and probably also the LED in order to save power(?). So if I for one reason or another would like to revert I'll just solder them back together. As for the Reed switch it seems that it should be soldered on the two outer holes right? Just got a bit confused what you meant by the picture.

[image]

PricelessToolkit, 2025-05-10:
1. Two outer holes or two right holes that I marked. 
2. To enable, solder them back!

You can keep LED enabled, it will not use much power.

closed 2025-05-10 by IDmedia